makerbase-mks/MKS-SERVO42B #6: driver overheating

lonelymyp, 2020-01-18:
If you set a current of 900 mA or higher, then the driver overheats already during the calibration process.

makerbase-mks, 2020-01-20:
Hi!
1. Increase the installation of heat sinks, and the heat dissipation will be further improved.
2. Does your motor need a load of 900ma?
3. Do you have any good suggestions for further optimization of SERVO42B.

lonelymyp, 2020-01-20:
3. The motor driver must be installed on the other side of the board. So that you can stick a small radiator to the heat sink area. 
so for example, done in the drivers tmc2208 
[image]

makerbase-mks, 2020-01-21:
This seems pretty good.

wildblue, 2020-01-24:
Put a small thermal compound pad between the motor and the pcb, right below the driver-ic. This should dissipate the heat in the aluminium body of the servo.

makerbase-mks, 2020-02-29:
We will consider this.

closed 2020-03-03 by makerbase-mks

dzid26, 2020-08-08:
Remove this [1500 offset](https://github.com/makerbase-mks/MKS-SERVO42B/blob/5328f04baca6f2080d3ea97f9990e5f72c244413/firmware/src/BSP/A4950.c#L122) that is present on both coils at all times. It reduces torque and adds heat. 
What was the point of the offset?!!

lonelymyp, 2020-08-08:
@dzid26 
> What was the point of the offset?!!
This bug appeared when they copied the code of the original nano stepper.
If you want it to work correctly, you need to change the code like the original.
https://github.com/Misfittech/nano_stepper/blob/master/firmware/stepper_nano_zero/A5995.cpp

dzid26, 2020-08-10:
It worked even with the offset, but slightly less torque and more heat. I just removed the offset and it runs cooler has more torque.  

Apart from that, I don't see any fundamental issues with the implementation.