makerbase-mks/MKS-SERVO42B #7: Incorrect current setting in the menu

lonelymyp, 2020-01-24:
Confused the name of the operating current and the holding current.

If a current of 100 mA is selected in the menu, a holding current of 100 mA is written to the memory. And the operating current is 200 mA.
[image]

Menu items are incorrectly renamed.
[image]

makerbase-mks, 2020-02-29:
Ok, thanks for your feedback, we will check it and confirm ...

lonelymyp, 2020-02-29:
Why is there no current? If I set the hold current to 200 mA and the operating current to 500 mA, then the motor is always energized for at least 200 mA. 
If the shaft is rotated, feedback is turned on and high current returns the shaft to its place.

dzid26, 2020-11-23:
Yes, it was very confusing. I changed that settings menu.  
https://github.com/dzid26/MKS-SERVO42B/commit/943960d989040feb27566cbd7193a4c57d3e349e