makerbase-mks/MKS-SERVO42B #8: not enough piece of code

lonelymyp, 2020-01-25:
when copying the original code 
an important piece of code was lost in the procedure
 `int32_t A4950_move(int32_t stepAngle, uint32_t mA)`
in file A4950.c

```
//scale sine result by current(mA)
dacSin=((int32_t)mA*(int64_t)abs(sin))/SINE_MAX;
//scale cosine result by current(mA)
dacCos=((int32_t)mA*(int64_t)abs(cos))/SINE_MAX;
```

makerbase-mks, 2020-02-29:
I have open file, and check it.
[image]

lonelymyp, 2020-02-29:
When you copied the code from the original project, you forgot to copy an important piece of code.
[image]

Without this piece, the pid algorithm works poorly. position correction current is not proportional and accuracy drops

dzid26, 2020-08-10:
What important piece is missing? Scaling down by SINE_MAX ?

martinbogo, 2020-08-25:
Hey -- did you make a PR from this issue?    I suggest you edit the code, allow the fork, then submit the PR.  AFTER you test the code.

dzid26, 2020-08-25:
@martinbogo What is not working?

Quas7, 2020-10-02:
Just to clarify, what I think is noted above.

In the original code from misfittech, the dacCos is scaled two times. 
In the MKS code, it is only scaled to the max current of 3300 instead.

code from misfit:
	_//scale sine result by current(mA)
	dacSin=((int32_t)mA*(int64_t)abs(sin))/SINE_MAX;
	//scale cosine result by current(mA)
	dacCos=((int32_t)mA*(int64_t)abs(cos))/SINE_MAX;_

	//convert value into DAC scaled to 3300mA max
	dacCos=(int32_t)((int64_t)dacCos*(DAC_MAX))/3300;
        //convert value into DAC scaled to 3300mA max
	dacSin=(int32_t)((int64_t)dacSin*(DAC_MAX))/3300;

Quas7, 2020-10-03:
Looking at the sineTable in sine.c I now wonder, why it is limited to 511 (~9bit) as there is a 12bit input from the A1333 encoder and the PWM to the A4950 also has a higher resolution.

Misfittech also has more resolution in the sineTable(s):
https://github.com/Misfittech/nano_stepper/blob/master/firmware/stepper_nano_zero/sine.cpp

dzid26, 2020-10-04:
@Quas7   The effective resolution is limited by timer max frequency 72Mhz. If it counts to 32767, it will be updated every  72Mhz/32767= ~2kHz. Slow.

With 512, it is ~140kHz. 
Now, the main loop by default runs at 8kHz (although I run it at 16kHz, for slightly smoother operation), so the timer should be at least as fast as the main loop. 2kHz is too slow.  
Going faster than the main loop should reduce latency because you wait very little time for the timer to trigger after setting a value. 
You would think 140kHz is overkill, but on the other hand, why would we need more precise current. It's not like the drivers are following the current perfectly anyway. 
Also remember, there is low pass filter (capacitor and resistor) that convert PWM into smooth voltage reference, So the faster the frequency, the smoother the voltage ref.

So I tried 32767 and the motor was sluggish (not catching up with main loop) and noisy (too slow for the low pass filter).
Then I tried 1023, and couldn't tell a difference between 511.
Then I tried 255 and it actually made a little less hissing than it was with 511 or 1023. 
Then I tried 127 and the high pitch noise disappeared completely! I am going to stick to that because I don't feel any performance difference. It feels as smooth as on 511/1023.

Quas7, 2020-10-04:
@dzid26 thanks for the nice background informations.

But I got a bit confused where you found the RC filter for the motion PWM signals. 
Signals IN1,2,3,4 reflect the motion signals to the [A4954](https://www.allegromicro.com/-/media/files/datasheets/a4954-datasheet.ashx) stepper motor driver (wrongly named "A9450" in the schematic) and as far as I see, they are connected directly to between uC and the driver. 
[image](https://user-images.githubusercontent.com/1175755/95029572-63e31c00-06a9-11eb-8d21-0edf3d326a5f.png)
[image]

Or did you maybe mean the VREF signals as there is actually an RC low pass in the schematic?
But VREF is only limiting the max current of the drivers and the sine table should not impact there (I hope...).

_**VREF**
The maximum value of current limiting is set by the selection of RSx and the voltage at the VREFx pin in each channel. [...]_ 
(And another topic but the VREF function likely has a flaw, as I see max current limits breached during calibration/re-test - it is obiously getting burning hot and clipping my bench supply current limit of 2A @15V.)

Anyway, I agree that we would have to stay out of the audible spectrum with >25kHz for the stepper PWM.
And I also agree, that even when using 128-ustepping we should still be fine with a 512-sine table (4x resolution) with regards to the microstep error as there are much bigger sources for errors than quantization in the complete system. ;)

dzid26, 2020-10-05:
@Quas7  
I have learnt that the sine wave is **indeed** controlled via **Vref**. 
IN1,2,3,4 is only used to select the phase. 

So the current control loop is really delegated to the A4954 chip. That is a little different from what you would normally see in FOC controllers, where MCU controls the current according to sensed current (which here it is not present, as there is no op-amps to measure shunt voltage). Surprisingly it works really well.  

I imagine that the only time the real current would be far from the requested current is at high speed - when BEMF prevents current flow. Unfortunately, we wouldn't know that this is occurring, so it's likely that in this case the current distribution between phases is not optimal. (Perhaps BEMF could be modeled to compensate for that).

Otherwise, especially at low speed, the "current limit" method for shaping the current signal works nicely.

Quas7, 2020-10-06:
Thanks again for the nice details of this concept! That explains, why I could net get my head around the datasheet. ;)

Regarding the overcurrent. 
I observed this only during calibration and, if I remember correctly, also in cal-testing. 
I just assume, that calibration is done in open-loop and somewhere the Vref limit is getting ignored or set to 3300mA max current limit.