makerbase-mks/MKS-SERVO42B #11: PCB change suggestion

ljubomirb, 2020-04-29:
Did you consider freeing pins `PB8`and `PB9` of `STM32`, currently used by _A9450_ (move it to other pins) and then add and connect _MCP2561-E/SN_ to it? I did not try this, but this particular _STM32_ has _CAN_ there, all it needs is 3.3V tranceiver. This would raise price per board maybe 1USD max, but would allow _CANBUS_ functionality. And that would be something. Think of connecting multiple motors in series, having simplified MB and such...
You may not even bother to add software support for it right away, just allow HW to have it as soon as possible, and include all libraries. The protocol can always be left undefined, so people can make do their own if they feel so.

dzid26, 2020-06-16:
I think it's even easier because CAN is also on PA11 and PA12. 
[image]
They seem to be unused:
[image]
Pin 31/32:
[image]
It's rotated 180deg, so they should be here:
[image]
But at the photo it seems it's connected to something:
[image]

Can someone double-check if I am correct and where is that path going on PCB?

dzid26, 2020-08-11:
I did. I added tja1050 transciever to the board. I am not recommending repeating that after me, as it was difficult. 
[image]
[image]
But anyways, it works. If anyone decides to build a board with the transciever, here a code

Quas7, 2020-11-15:
for everybody, who sumbles over this issue: the latest S42B v2 has a CAN interface now.