makerbase-mks/MKS-SERVO42B #12: PID Tuning

haudamekki, 2020-05-13:
How could I tune the PID values? Is there a serial interface like the MisfitTech Smart Stepper has?

lonelymyp, 2020-06-13:
@haudamekki 
 No. This is a bad copy of the MisfitTech old version, there is a lack of many necessary things.

dzid26, 2020-06-20:
> @haudamekki
> No. This is a bad copy of the MisfitTech old version, there is a lack of many necessary things.

What is it lacking apart from serial?
I think more similar copy is atmel based Servo42A https://github.com/makerbase-mks/MKS-SERVO42A

lonelymyp, 2020-10-05:
> What is it lacking apart from serial?

For example, the code for control on uart, [commands.cpp](https://github.com/Misfittech/nano_stepper/blob/master/firmware/stepper_nano_zero/commands.cpp) cut out completely.

The code for different modes of PID operation was ignored. 
There are 3 different modes in the original, SIMPLE PID, VELOCITY PID, POSITIONAL PID.
Only 1 mode was left here, SIMPLE PID.

Cut-out holding current control. The holding current cannot be set in the menu, it is rigidly tied to the operating current.