makerbase-mks/MKS-SERVO42B #13: Marlin configuration

calabia2000, 2020-05-13:
I have bought this article (it has not reached me yet), in marlin 2.0 you must define the type of driver used (a4988, tmcxxx, etc).  Using this servo42b what type of driver should I define in marlin 2.0

makerbase-mks, 2020-05-18:
You don't need to define it, you can use it directly.

closed 2020-05-18 by makerbase-mks

ghent360, 2020-05-21:
Use the default a4988 driver with no special current control. That is what the previous comment meant - you don't have to change the driver the default just works.

You have to match the microsteps in the Marlin configuration with the servo configuration. I believe the default is 16 on both.

3DMadeUp, 2021-10-04:
Sorry, i buy the mks servo42b but i have more noise on y axis when change direction........why??
Thanks for help, i just try the marlin configuration