makerbase-mks/MKS-SERVO42B #15: Not Open Source

npgoalkeeper, 2020-07-15:
PDF's of schematic files are not source.

dzid26, 2020-08-05:
I Agree.   
The life would be much easier for some DIYers if they wanted to improve the PCB. 👍

martinbogo, 2020-08-25:
I'm working on re-entering the schematics into KiCAD -- the resulting PCB/Gerber won't precisely match what MKS has done though. 

MKS -- would you please release the schematic/board files in whatever platform you are using?

JustGitting, 2020-12-11:
@npgoalkeeper Agreed!

@martinbogo You may already have seen this, but [pdf2gerb](https://swannman.github.io/pdf2gerb/) might be of help. 

@makerbase-mks Please release the gerber files so others can improve on your designs and your hard work is preserved and protected (win-win)