makerbase-mks/MKS-SERVO42B #17: Update wiki with ST-link isntruction to disable "Read Out Protection"

dzid26, 2020-08-04:
Error when trying to connect to STM32:
```
Can not read memory!
Disable Read Out Protection and retry.
```

1. In order to unprotect the chip with [ST-LINK](https://www.st.com/en/development-tools/stsw-link004.html), one has to click _Target_ -> _Option Bytes_:
-> change **_Read-Out Protection_** - **Disabled**. 
[image]

2. Then it is possible to switch to System memory (0x1FFFF800) and see firmware binary:
[image]

3. Flashing is possible now.

makerbase-mks, 2020-08-05:
Hi! You need Unsecure chip by J flash or other tool, but i know't your tool hold Unsecure chip function.
[image]

dzid26, 2020-08-05:
I know.
I provided solution for ST-link, because it was not obvious what to do. 
You can add my instruction to the wiki https://github.com/makerbase-mks/MKS-SERVO42B/wiki/MKS-SERVO42B-User-Manual#how-to-upload-firmware so it's easier for people understand how to do it with ST-Link tool.

mustafatrk0, 2021-10-17:
In section Read Out Protection, it says Level0, Level1, Level2.
How can I turn off Read Out Protection

dzid26, 2021-10-17:
> In section Read Out Protection, it says Level0, Level1, Level2. How can I turn off Read Out Protection
This doesn't work for you?
https://github.com/makerbase-mks/MKS-SERVO42B/issues/17#issue-672503790

fpvmate, 2021-12-11:
> unfortunately not working. When I try to change from Level 1 to Level 0 I get error "Could not set Options byte! Please reset the target and retry"

I also get same error did anyone solved this ?

dzid26, 2021-12-11:
> > unfortunately not working. When I try to change from Level 1 to Level 0 I get error "Could not set Options byte! Please reset the target and retry"
> 
> I also get same error did anyone solved this ?

The first post says what to do.

fpvmate, 2021-12-11:
> > > unfortunately not working. When I try to change from Level 1 to Level 0 I get error "Could not set Options byte! Please reset the target and retry"
> > 
> > 
> > I also get same error did anyone solved this ?
> 
> The first post says what to do.

I tried that and it didn't work that's why asked again.

up2smoke, 2022-02-04:
+1 disable protection is nt possible

dzid26, 2022-03-09:
> In section Read Out Protection, it says Level0, Level1, Level2. How can I turn off Read Out Protection
I had a look at it again. F103 series chip doesn't have protection _levels_. Only _disabled/enabled_ options. So you are writing in the wrong Github issue or maybe somehow you were debugging wrong target.
This should be reported device:
[image]

fpvmate, 2022-03-09:
> In section Read Out Protection, it says Level0, Level1, Level2. How can I turn off Read Out Protection
> I had a look at it again. F103 series chip doesn't have protection _levels_. Only [...]

But what about stm32f405 chips in that while I was using in my flight controller the board stopped responding and when i connect the processor with st-link it's saying read out protection is enabled and when i try to make it to level 0 it's not happening and says failed in between . Any solution for that or is it possible to erase the entire chip ? Can anything be done with the help of boot pins of the processor?

dzid26, 2022-03-13:
If it is set to Level2 then it is irreversible. https://www.st.com/resource/en/application_note/an4701-proprietary-code-readout-protection-on-microcontrollers-of-the-stm32f4-series-stmicroelectronics.pdf 
Anyway, your problem doesn't belong to this issue.

closed 2022-03-13 by dzid26

Sasha7b9Work, 2022-08-16:
This helped me: https://community.st.com/s/question/0D50X00009Xkdy5SAB/not-able-to-update-option-bytes
(update firmware  ST-Link utility)

huanbaose, 2022-09-05:
> 在读取保护部分中，它显示 Level0、Level1、Level2。 如何关闭读出保护

You need to select Level0.