makerbase-mks/MKS-SERVO42B #32: Can it be used with 0.9 stepper?

HanSolo30, 2021-09-28:
I have a couple of 0.9 degrees steppers and I want to test it

mpact1984, 2021-09-28:
Hi HanSolo30. 

I didn’t have one here by me at the moment, but as far as i can remember, if you have also the Display in the menu you can choose how many steps have to be performed for one complete revolution. 

Best regards

HanSolo30, 2021-09-28:
Yes, but I always have an error when I try to calibrate it.
I assume that when the drivers waits for a 1.8 degrees in a step, only gets 0.9 and that's why not working.
Maybe need some firmware modification.

mpact1984, 2021-09-28:
Hmmm … a quick firmware check on Github seems that it should work.

Do you connect it to 5V or 3.3V?

HanSolo30, 2021-09-28:
Good question. It is connected straight to a MKS Gen L.
With a 1.8 degrees steppers works fine.

HanSolo30, 2021-09-29:
I found this in the file MKS.c

options_t stepOptions[] = {
		{"200"},
		{"400"},
		{""},
};

//returns the index of the stepOptions when called
// with no arguments.
int motorSteps(int argc, char *argv[])
{
	if (argc == 0)
	{
		uint16_t i;
		i = NVM->motorParams.fullStepsPerRotation;
		if (i == 400)
		{
			return 1;
		}
		return 0; //default to 200
	}
	if (argc > 0)
	{
		uint16_t i;
		nvm_t params;
		memcpy((void *)&params, (void *)NVM, sizeof(params));
		i = atol(argv[0]);
		if (i != params.motorParams.fullStepsPerRotation)
		{
			params.motorParams.fullStepsPerRotation = i;
			nvmWriteConfParms(&params);
		}
	}
	return 0;
}

Where the board "knows" that I have a 400 full steps motor? every time I run the calibration and test, I get a 90° error.

HanSolo30, 2021-09-29:
Well, I understand that, in the first power on, the board "read" de kind of motor and store in the NVM. That test it runs once.
Maybe my solution is to clear the NVM.

Is there a way to do that?